Sienna-Platform/PowerSimulationsDynamics.jl #49: Minor correction in documentation about Network Equations in https://nrel-siip.github.io/PowerSimulationsDynamics.jl/stable/component_models/network/

amar-iastate, 2021-01-15:
**Is your feature request related to a problem? Please describe.**
This is related to the equations in documentation at https://nrel-siip.github.io/PowerSimulationsDynamics.jl/stable/component_models/network/
In power system literature, the complex quantities are represented by Capital alphabets. The equations (1), (6), (7) & (8) need to be updated. Also equation (1) should be `0 = i(x,v) - Y*v` and not `0 = i(x,v) - Y*x`.

kdheepak, 2021-01-15:
Are the quantities represented here instantaneous?

rodrigomha, 2021-01-16:
> Are the quantities represented here instantaneous?

No, they are phasors.

@amar-iastate Although I agree with the notation being Capital, I was looking here to differentiate matrices with vectors using that type of notation here. I will discuss a good strategy to avoid confusion here.

amar-iastate, 2021-01-16:
No.. The quantities are meant to be the phasor representation. The way it is written looks like it is the instantaneous version, which is not the case.

amar-iastate, 2021-01-16:
> > Are the quantities represented here instantaneous?
> 
> No, they are phasors.
> 
> @amar-iastate Although I agree with the notation being Capital, I was looking here to differentiate matrices with vectors using that type of notation here. I will discuss a good strategy to avoid confusion here.

I think matrices are usually bold capitals and vectors are bold small alphabets. Equations (2) - (5) seem to be written in the normal convention.

PR #57 (2021-05-07): Fix some issues and update docs
The following PR closes #49 and #51.

In addition, add documentation for `FixedFrequency` PLL and `ActivePowerDroop` outer control model.