Sienna-Platform/PowerSimulationsDynamics.jl #50: Initialization failure with dynamic lines

m-bossart, 2021-03-22:
Initialization for a system with all dynamic lines (ie no static lines) fails to converge to a stable equilibrium. The same system with all static lines converges, and  the dynamic/static nature of the lines should not change the steady state solution.

jd-lara, 2021-03-23:
@m-bossart thanks, do you have a case we can use to reproduce?

m-bossart, 2021-03-24:
This is the `02_line_dynamics.jl` example script with all of the lines made dynamic instead of just one, which reproduces what I see. 
```
# Line Modeling simulation with [PowerSimulationsDynamics.jl](https://github.com/NREL-SIIP/PowerSimulationsDynamics.jl)

# **Originally Contributed by**: Rodrigo Henriquez and José Daniel Lara

# ## Introduction

# This tutorial will introduce an example of considering dynamic lines in `PowerSimulationsDynamics`.
# Note that this tutorial is for `PowerSimulationsDynamics`.

# This tutorial presents a simulation of a three-bus system, with an infinite bus (represented as a
# voltage source behind an impedance) at bus 1, a one d- one q- machine on bus 2 and an inverter
# of 19 states, as a virtual synchronous machine at bus 3. The perturbation will be the trip of
# two of the three circuits (triplicating its resistance and impedance) of the line that connects
# bus 1 and bus 3. This case also consider a dynamic line model for connection between buses
# 2 and 3. We will compare it against a system without dynamic lines.

# In addition, note that `PowerSimulationsDynamics` will convert ConstantPower loads to RLC
# loads for transient simulations.

# It is recommended to check `Tutorial 1: OMIB` first, since that includes more details and
# explanations on all definitions and functions.

# # Step 1: Package Initialization
using SIIPExamples
using PowerSimulationsDynamics
PSID = PowerSimulationsDynamics
using PowerSystems
using Sundials
using Plots

# # Step 2: Data creation
file_dir = joinpath(
    dirname(dirname(pathof(SIIPExamples))),
    "script",
    "4_PowerSimulationsDynamics_examples",
    "Data",
)
threebus_sys = System(joinpath(file_dir, "threebus_sys.json"));

# In addition, we will create a new copy of the system on which we will simulate the same
# case, but will consider dynamic lines:

threebus_sys_dyn = deepcopy(threebus_sys);

# # Step 3: Create the fault and simulation on the Static Lines system

# First, we construct the perturbation, by properly computing the new Ybus on the system:

#Make a copy of the original system
sys2 = deepcopy(threebus_sys)
#Triplicates the impedance of the line named "BUS 1-BUS 3-i_1"
fault_branches = get_components(ACBranch, sys2)
for br in fault_branches
    if get_name(br) == "BUS 1-BUS 3-i_1"
        br.r = 3 * br.r
        br.x = 3 * br.x
        b_new = (from = br.b.from / 3, to = br.b.to / 3)
        br.b = b_new
    end
end
#Obtain the new Ybus
Ybus_fault = Ybus(sys2).data
#Define Fault: Change of YBus
Ybus_change = NetworkSwitch(
    1.0, #change at t = 1.0
    Ybus_fault, #New YBus
);

[... 103 lines cut ...]
print_device_states(sim_dyn)
#Will export a dictionary with the initial condition values to explore
x0_init_dyn = PSID.get_initial_conditions(sim_dyn)

# # Step 5.1: Store the solution

series2_dyn = get_voltagemag_series(sim_dyn, 102)
zoom_dyn = [
    (series2_dyn[1][ix], series2_dyn[2][ix]) for
    (ix, s) in enumerate(series2_dyn[1]) if (s > 0.90 && s < 1.6)
];

# # Step 6.1: Compare the solutions:

# We can observe the effect of Dynamic Lines

Plots.plot(series2_dyn, label = "V_gen_dyn")
Plots.plot!(series2, label = "V_gen_st", xlabel = "Time [s]", ylabel = "Voltage [pu]")

# that looks quite similar. The differences can be observed in the zoom plot:

Plots.plot(zoom_dyn, label = "V_gen_dyn")
Plots.plot!(zoom, label = "V_gen_st", xlabel = "Time [s]", ylabel = "Voltage [pu]")

```

PR #59 (2021-05-12): fix branch bug
closes #48
closes #50
closed 2021-05-12 by jd-lara