Sienna-Platform/PowerSimulationsDynamics.jl #60: Add Frequency Reference Options for Simulation

rodrigomha, 2021-05-12:
Including Center-of-Inertia for system frequency, or using Frequency-Divider approaches to obtain frequency for each bus that can be passed directly to inverters (skipping PLL implementation).

m-bossart, 2022-10-13:
@jd-lara Would this change also allow for using a device as the frequency ` ReferenceBus` that doesn't have frequency as a dynamic state (Droop GFM)?

jd-lara, 2022-10-13:
We can account for that as an option. This is an interesting question about how to implement it. We can add another option.

m-bossart, 2022-10-13:
Yeah it seems like the current implementation is limited to devices that have frequency as a state that is therefore already stored in the result.