Sienna-Platform/PowerSimulationsDynamics.jl #118: Refactor inner vars calculations

jd-lara, 2021-09-02:
Currently, the inner vars are updated inside each component model function. This leads to some sequencing issues with outdated variables as demonstrated in AC1A and REECB. The inner vars update strategy needs updating to handle those cases.

rodrigomha, 2021-09-02:
So my main idea is to refactor our `device.jl` file, on which after reading the states and general system parameters, do an `update_inner_vars(device_states, dynamic_device, ...)` that has multiple calls of `update_voltages_vars(...)`, `update_machine_vars(...)`, `update_avr_vars(...)`, etc. Then when we call the `mdl_component_ode(...)` we do not update any inner_vars, and simply update the ODEs.

The main issue will be that we will repeat some computations, but I think is worth to get our code more ordered. Another issue is that now we require users to develop their own models to have an `update_function` for the inner vars.

https://github.com/NREL-SIIP/PowerSimulationsDynamics.jl/blob/8fad343de86a32db86e701829b295e6d7870225a/src/models/device.jl#L34-L77

jd-lara, 2021-09-23:
Currently only implemented for some models that require it. Not a comprehensive fix yet.