Sienna-Platform/PowerSimulationsDynamics.jl #168: Change from DiffEqBase to SciMLBase

ChrisRackauckas, 2021-12-08:
DiffEqBase is the lowest common denominator for the DiffEq packages, not necessarily the whole SciML ecosystem, and so it has a lot DiffEq dependencies. These are generally not required by downstream packages. If what you're looking for is a way to define problems without having most dependencies, we recommend you use SciMLBase as the dependency since everything like ODEProblem, SteadyStateProblem, etc. is defined there. We basically recommend depending on SciMLBase for problem definitions, and solver packages for specific solvers, but generally most non-SciML packages should not be depending on DiffEqBase directly (given the split of SciMLBase in 2021)

For more details see: https://diffeq.sciml.ai/stable/features/low_dep/ and https://discourse.julialang.org/t/psa-the-right-dependency-to-reduce-from-differentialequations-jl/72757

Let me know if you need any help updating this, though for almost all dependents here it should be a trivial name change as you're actually using pieces from SciMLBase.

jd-lara, 2022-01-20:
@ChrisRackauckas thanks for the note. We can't remove DiffEqBase since Callbacks and CallbackSet are defined in DiffEqBase and not in SciMLBase

closed 2022-01-20 by jd-lara

ChrisRackauckas, 2022-01-20:
Callbacks and CallbackSet are defined in SciMLBase. https://github.com/SciML/SciMLBase.jl/blob/master/src/callbacks.jl

jd-lara, 2022-01-21:
fixed via #175