Sienna-Platform/PowerSimulationsDynamics.jl #181: read_results is not defined

mkaracelebi, 2022-02-01:
read_results( ) function is not defined in the latest version.

jd-lara, 2022-02-01:
This issue makes me think again that you are using the incorrect version of PowerSimulationsDynamics. The function `read_results` was introduced in v0.8.5.

mkaracelebi, 2022-02-01:
Thanks Jose, indeed my version was off but it is quite strange since I have imported all packages this week. Why did Julia import an older version instead of the newest?

closed 2022-02-01 by mkaracelebi

jd-lara, 2022-02-01:
sometimes the package manager will resolve to older versions if the dependency constraints allow it.

kalpaniosu, 2023-09-12:
Can someone provide the read_results function code. I cannot find it in my version

rodrigomha, 2023-09-13:
> Can someone provide the read_results function code. I cannot find it in my version

It simply returns the SimulationResults struct. See here: https://github.com/NREL-Sienna/PowerSimulationsDynamics.jl/blob/012e68c55b61f8aca6493c7fc7ae422ebd48bc90/src/base/simulation.jl#L555

The SimulationResults struct is defined here:
https://github.com/NREL-Sienna/PowerSimulationsDynamics.jl/blob/012e68c55b61f8aca6493c7fc7ae422ebd48bc90/src/base/simulation_results.jl

kalpaniosu, 2023-09-13:
> It simply returns the SimulationResults struct. See here:
>
> https://github.com/NREL-Sienna/PowerSimulationsDynamics.jl/blob/012e68c55b61f8aca6493c7fc7ae422ebd48bc90/src/base/simulation.jl#L555
>
> The [...]

Thank you so much for your reply.Appreciate it.