Sienna-Platform/PowerSimulationsDynamics.jl #257: ERROR: LoadError: UndefVarError: BranchTrip not defined

frostyduck, 2022-07-15:
I want to run this code

```
using PowerSimulationsDynamics
using PowerSystems
using Logging
using Sundials
using Plots

PSD = PowerSimulationsDynamics
                
new_sys = System(joinpath("14bus.json"))
time_span = (0.0, 30.0)
perturbation_trip =BranchTrip(1.0, Line, "BUS 02-BUS 04-i_4")
sim = PSD.Simulation(ResidualModel,  new_sys,  time_span,)
```

Howewer I get the following error:

**ERROR: LoadError: UndefVarError: BranchTrip not defined**

jd-lara, 2022-07-15:
Can you provide details about which version of PowerSimulationsDynamics are you running? 

It looks like you might be running an older version of the package.

frostyduck, 2022-07-15:
Version 1.7

jd-lara, 2022-07-15:
@frostyduck which version of PowerSimulationsDynamics.jl?

rodrigomha, 2022-07-15:
You can try to use `] st` to see your version. If you wanna update do `] up` to enter the pkg environment and update your packages

frostyduck, 2022-07-16:
@jd-lara sorry, PowerSimulationsDynamics v0.1.1

frostyduck, 2022-07-16:
@rodrigomha thank you, I have looked my version (v.0.1.1) and after that I tried to use `] up`, but was updated only some packages, excluding PowerSimulation Dynamics.jl

jd-lara, 2022-07-16:
@frostyduck can you share the output of doing 

```
] st
```

It seems that you have some dependencies that fixed PowerSimulationsDynamics to a very old version.

frostyduck, 2022-07-17:
@jd-lara @rodrigomha 
I have found the problem. The LITS.jl package disturbed the installation the latest version Power Simulation Dynamics.jl. Thank for your help!

closed 2022-07-17 by jd-lara