Sienna-Platform/PowerSimulationsDynamics.jl #300: Problem to get active power series in DERA

JorgeS23, 2022-12-04:
Hi
I've been having problems with the `get_activepower_series` function for the dera model. It gives me the next error:

`ERROR: LoadError: MethodError: no method matching compute_output_current(::SimulationResults, ::AggregateDistributedGenerationA, ::Vector{Float64}, ::Vector{Float64}, ::Nothing)
Closest candidates are:
  compute_output_current(::SimulationResults, ::G, ::Vector{Float64}, ::Vector{Float64}, ::Union{Nothing, Float64}) where G<:DynamicGenerator at C:\Users\jlsan\.julia\packages\PowerSimulationsDynamics\Al2pp\src\post_processing\post_proc_generator.jl:6
  compute_output_current(::SimulationResults, ::G, ::Vector{Float64}, ::Vector{Float64}, ::Union{Nothing, Float64}) where G<:DynamicInverter at C:\Users\jlsan\.julia\packages\PowerSimulationsDynamics\Al2pp\src\post_processing\post_proc_inverter.jl:6
  compute_output_current(::SimulationResults, ::SingleCageInductionMachine, ::Vector{Float64}, ::Vector{Float64}, ::Union{Nothing, Float64}) at C:\Users\jlsan\.julia\packages\PowerSimulationsDynamics\Al2pp\src\post_processing\post_proc_loads.jl:6`

I think the problem is in the DERA definition in PowerSystems.jl, since the AggregateDistributedGenerationA function is a subtype of DynamicInjection and there is no function to compute the output current for devices of that subtype. There is such a function for devices of subtype DynamicInverter and DynamicGenerator (both subtypes of DynamicInjection), but not for DynamicInjection.

rodrigomha, 2022-12-04:
Thanks for catching this. You are probably right.

I can fix this ASAP before the next release. Once I pushed this fix to master I'll update this thread.

jd-lara, 2022-12-05:
@rodrigomha, we should also implement a more graceful error while we do the push for the post-processing of results.

jd-lara, 2022-12-19:
closed via #302

closed 2022-12-19 by jd-lara