Sienna-Platform/PowerSimulationsDynamics.jl #315: Update equation formatting in markdown files under component_models

HaleyRoss, 2023-04-10:
**Describe the bug**
When viewing markdown files using the "preview" feature in VSCode, some equations format correctly and some do not. This is observed in avr.md, converter.md, filters.md, freq_esti.md, inner_control.md, loads.md, machines.md, outer_control.md, pss.md, shafts.md, and turbine_gov.md

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: macOS 11.7.4
 - Browser: safari
 - Version: 16.3.1

**Additional context**
Math+Markdown installed and 'julia' used as the delimiter setting. Using VSCode version 1.77.1.

rodrigomha, 2023-04-10:
I'm not familiar with how VSCode handles the mathematical expressions, but I think this is related with the mathengine being used in VSCode preview.

When we built the docs we need to select MathJax as mathengine (see here: https://github.com/NREL-SIIP/PowerSimulationsDynamics.jl/blob/21fe4bad736d908fc1c76e2a2e91dae972a56f51/docs/make.jl#L5). 

This is mentioned in the past here: https://stackoverflow.com/questions/67562772/vscode-markdown-preview-not-supporting-some-math-commands. It may be that installing some extensions like Markdown Enhanced could make it work, but I'm not sure. I'll leave the issue open for now, and let me know if you can find some solution on your side.

In any case, you can always check the docs online to get the equations working properly: https://nrel-siip.github.io/PowerSimulationsDynamics.jl/stable/component_models/avr/

rodrigomha, 2023-04-12:
Just to close this issue for now. I've been looking around and it seems that align is particularly an issue when using this and it is somewhat recommended using the environment aligned instead of align.

I may take a look in the future of replacing our docs to use aligned.

closed 2023-04-12 by rodrigomha