Sienna-Platform/PowerSimulationsDynamics.jl #342: Fix NetworkSwitch perturbation tutorial

rodrigomha, 2023-07-10:
We need to explicitly state to use `using PowerNetworkMatrices` when creating Ybus. 

In addition, we should discuss if [NetworkSwitch](https://nrel-sienna.github.io/PowerSimulationsDynamics.jl/stable/api/public/#PowerSimulationsDynamics.NetworkSwitch) should ONLY use a SparseMatrix or not. From a user perspective we should allow the user to use directly a dense matrix if they want.

In any case, we need to update the tutorial of Perturbations since it is using Dense Matrices: https://nrel-sienna.github.io/PowerSimulationsDynamics.jl/stable/perturbations/

jd-lara, 2023-07-11:
I don't think this is a bug. We have always assumed that the user would pass a Bus object which is by default sparse.

rodrigomha, 2026-11-06:
Fixed in #349: the perturbations page now converts to sparse before NetworkSwitch, and NetworkSwitch also accepts a PNM.Ybus. Dense input stays unsupported by design. Closing.

closed 2026-11-06 by rodrigomha