Sienna-Platform/PowerSimulationsDynamics.jl #356: Typos in documents: LCL Filter state-space model reference frame should be on the dq0 axis.

XingyuCiao, 2023-11-15:
Hi! The correct model should be:
[image]

rodrigomha, 2023-11-15:
Hi, is the only difference that we use R and I instead of DQ? This is just a convention that we use to refer to the reference frame of the network as RI instead of the internal reference frames of dq used for devices.

XingyuCiao, 2023-11-15:
> Hi, is the only difference that we use R and I instead of DQ? This is just a convention that we use to refer to the reference frame of the network as RI instead of the internal reference frames of dq used for devices.
Hi, that makes sense to me. 
Just one more small typo, for eq(1e) and (1f), the filter current i_r and i_i should be used instead of converter current i_r,cv and i_i,cv.

rodrigomha, 2023-11-15:
> Hi, that makes sense to me. Just one more small typo, for eq(1e) and (1f), the filter current i_r and i_i should be used instead of converter current i_r,cv and i_i,cv.

Thanks for this. I'll fix this typo in the next release!

jd-lara, 2024-01-05:
@rodrigomha can we fix this for the 0.13 release?

PR #361 (2024-01-05): update docs
closes #356
closed 2024-01-05 by jd-lara