Sienna-Platform/PowerSimulationsDynamics.jl #396: Simulate TAMU 37-bus system

Robbybp, 2025-02-14:
**Is your feature request related to a problem? Please describe.**
I'd like to use this package to run simulations of the 37-bus synthetic grid provided by Texas A&M, available for download here: https://electricgrids.engr.tamu.edu/hawaii40/. I am using the .raw and .dyr files provided in this dataset. I am preprocessing the .dyr file with the following sed command because REGCA1 generators don't seem to be supported by PowerSystems (see https://github.com/NREL-Sienna/PowerSystems.jl/issues/1257):
```
sed -i s/REGCA1/REECB1/g Hawaii40_20231026.dyr
```

I run the following code:
```julia
using PowerSystems
using PowerSimulationsDynamics
rawfile = "Hawaii40_20231026.RAW"
dyrfile = "Hawaii40_20231026.dyr"
sys = System(rawfile, dyrfile)
sim = Simulation(ResidualModel, sys, pwd(), (0.0, 30.0), initialize_simulation = false, PowerSimulationsDynamics.Perturbation[])
```
And get an error:
```console
┌ Error: ResidualModel failed to build                                                                                                                                                                                         
│   exception =                                                                                                                                                                                                                
│    MethodError: no method matching mdl_tg_ode!(::SubArray{ForwardDiff.Dual{ForwardDiff.Tag{PowerSimulationsDynamics.var"#74#76"{PowerSimulationsDynamics.SystemModel{ResidualModel, PowerSimulationsDynamics.NoDelays, PowerS
│    The function `mdl_tg_ode!` exists, but no method is defined for this combination of argument types.                                                                                                                       
│                                                                                                                                                                                                                              
│    Closest candidates are:                                                                                                                                                                                                   
│      mdl_tg_ode!(::AbstractArray{<:Union{Float64, ForwardDiff.Dual}}, ::AbstractArray{<:Union{Float64, ForwardDiff.Dual}}, ::AbstractArray{<:Union{Float64, ForwardDiff.Dual}}, ::Union{Float64, ForwardDiff.Dual}, ::PowerSi
│       @ PowerSimulationsDynamics ~/.julia/dev/PowerSimulationsDynamics/src/models/generator_models/tg_models.jl:655                                                                                                          
│      mdl_tg_ode!(::AbstractArray{<:Union{Float64, ForwardDiff.Dual}}, ::AbstractArray{<:Union{Float64, ForwardDiff.Dual}}, ::AbstractArray{<:Union{Float64, ForwardDiff.Dual}}, ::Union{Float64, ForwardDiff.Dual}, ::PowerSi
│       @ PowerSimulationsDynamics ~/.julia/dev/PowerSimulationsDynamics/src/models/generator_models/tg_models.jl:559                                                                                                          
│      mdl_tg_ode!(::AbstractArray{<:Union{Float64, ForwardDiff.Dual}}, ::AbstractArray{<:Union{Float64, ForwardDiff.Dual}}, ::AbstractArray{<:Union{Float64, ForwardDiff.Dual}}, ::Union{Float64, ForwardDiff.Dual}, ::PowerSi
│       @ PowerSimulationsDynamics ~/.julia/dev/PowerSimulationsDynamics/src/models/generator_models/tg_models.jl:472                                                                                                          
│      ...                                                                                                                                                                                                                     
│                                                                                                                                                                                                                              
│    Stacktrace:                                                                                                                                                                                                               
│      [1] device!(device_states::SubArray{ForwardDiff.Dual{ForwardDiff.Tag{PowerSimulationsDynamics.var"#74#76"{PowerSimulationsDynamics.SystemModel{ResidualModel, PowerSimulationsDynamics.NoDelays, PowerSimulationsDynamic
│        @ PowerSimulationsDynamics ~/.julia/dev/PowerSimulationsDynamics/src/models/device.jl:45                                                                                                                              
│      [2] system_residual!(out::Vector{ForwardDiff.Dual{ForwardDiff.Tag{PowerSimulationsDynamics.var"#74#76"{PowerSimulationsDynamics.SystemModel{ResidualModel, PowerSimulationsDynamics.NoDelays, PowerSimulationsDynamics.J
│        @ PowerSimulationsDynamics ~/.julia/dev/PowerSimulationsDynamics/src/models/system.jl:86                                                                                                                              
│      [3] SystemModel                                                                                                                                                                                                         
│        @ ~/.julia/dev/PowerSimulationsDynamics/src/models/system.jl:51 [inlined]                                                                                                                                             
│      [4] #74                                                                                                                                                                                                                 
│        @ ~/.julia/dev/PowerSimulationsDynamics/src/base/jacobian.jl:135 [inlined]                                                                                                                                            
│      [5] chunk_mode_jacobian!(result::Matrix{Float64}, f!::PowerSimulationsDynamics.var"#74#76"{PowerSimulationsDynamics.SystemModel{ResidualModel, PowerSimulationsDynamics.NoDelays, PowerSimulationsDynamics.JacobianCache
│        @ ForwardDiff ~/.julia/packages/ForwardDiff/UBbGT/src/jacobian.jl:187                                                                                                                                                 
│      [6] jacobian!                                                                                                                                                                                                           
│        @ ~/.julia/packages/ForwardDiff/UBbGT/src/jacobian.jl:84 [inlined]                                                                                                                                                    
│      [7] jacobian!                                                                                                                                                                                                           
│        @ ~/.julia/packages/ForwardDiff/UBbGT/src/jacobian.jl:79 [inlined]                                                                                                                                                    
│      [8] (::PowerSimulationsDynamics.var"#75#77"{ForwardDiff.JacobianConfig{ForwardDiff.Tag{PowerSimulationsDynamics.var"#74#76"{PowerSimulationsDynamics.SystemModel{ResidualModel, PowerSimulationsDynamics.NoDelays, Power
ardDiff.Dual{ForwardDiff.Tag{typeof(PowerSimulationsDynamics.system_residual!), Float64}, Float64, 12}}}, Int64}, Float64}, Float64, 12}}}}, PowerSimulationsDynamics.var"#74#76"{PowerSimulationsDynamics.SystemModel{Residual
│        @ PowerSimulationsDynamics ~/.julia/dev/PowerSimulationsDynamics/src/base/jacobian.jl:139                                                                                                                             
│      [9] PowerSimulationsDynamics.JacobianFunctionWrapper(m!::PowerSimulationsDynamics.SystemModel{ResidualModel, PowerSimulationsDynamics.NoDelays, PowerSimulationsDynamics.JacobianCache{typeof(PowerSimulationsDynamics.s
│        @ PowerSimulationsDynamics ~/.julia/dev/PowerSimulationsDynamics/src/base/jacobian.jl:148                                                                                                          [... cut ...]
I would like to be able to simulate (something resembling) this dataset. If there is some workaround I can use (e.g., another dataset modification), I would be happy to know about that as well.

Unfortunately, I am not an expert in power system dynamics and do not know what properties of the data files are causing `GeneralGovModel` to be used instead of one of the supported models. I assume the governor models are chosen based on the generator types in the .dyr file, and that this error is due to a disconnect between the generator models supported by PowerSystems and the turbine-governor models supported by PSID.

rodrigomha, 2025-02-14:
Yeah, we do not support GGOV1. The alternative usually here is modify the raw file directly, to add a supported governor. Usually using a GAST or TGOV is enough if your data is not that complicated.

If you don't want to modify directly the raw file, you need to load the system, and then remove the dynamic injector from the generator with GGOV1 and a new dynamic injector with the new data.

For example, here is a method that remove the governors from the system if you have a list of names of generators that you want to modify. You may want to create a different Governor rather than `TGFixed` for your case.

```
function remove_governors!(sys::System, gen_names::Vector{String})
    for name in gen_names
        gen = get_component(ThermalStandard, sys, name)
        dyn_gen = get_dynamic_injector(gen)
        new_tg = TGFixed(efficiency=1.0)
        new_gen_model = DynamicGenerator(
            name=get_name(gen),
            ω_ref=1.0,
            machine=deepcopy(get_machine(dyn_gen)),
            shaft=deepcopy(get_shaft(dyn_gen)),
            avr=deepcopy(get_avr(dyn_gen)),
            prime_mover=new_tg,
            pss=deepcopy(get_pss(dyn_gen)),
            base_power=get_base_power(dyn_gen),
        )
        set_dynamic_injector!(gen, nothing)
        remove_component!(sys, dyn_gen)
        add_component!(gen, new_gen_model, gen)
    end
end
```

Robbybp, 2025-02-14:
Thanks for the quick response! When I try this, I get the new error:
```console
ERROR: LoadError: MethodError: no method matching get_machine(::DynamicInverter{RenewableEnergyConverterTypeA, OuterControl{ActiveRenewableControllerAB, ReactiveRenewableControllerAB}, RECurrentControlB, FixedDCSource, Fixe
The function `get_machine` exists, but no method is defined for this combination of argument types.                                                                                                                            
                                                                                                                                                                                                                               
Closest candidates are:                                                                                                                                                                                                        
  get_machine(::PowerSimulationsDynamics.DynamicWrapper{T}) where T<:DynamicGenerator                                                                                                                                          
   @ PowerSimulationsDynamics ~/.julia/dev/PowerSimulationsDynamics/src/base/device_wrapper.jl:314                                                                                                                             
  get_machine(::DynamicGenerator)                                                                                                                                                                                              
   @ PowerSystems ~/.julia/dev/PowerSystems/src/models/dynamic_generator.jl:137                                                                                                                                                
```

I guess machine, shaft, etc. don't mean much for the renewable generators in this dataset? Rather than heavily modify this test case, I'll probably shift my focus to using one of the test cases in the PSID examples (unless you have an easy way to get around this as well).

rodrigomha, 2025-02-14:
It's probably a good idea to do that.

Regarding your problem, you need to filter when the generator is an inverter. In PSID, we parse inverters differently than machines. I'll recomend you to check the metamodels of inverter and generators in the documentation: https://nrel-sienna.github.io/PowerSimulationsDynamics.jl/stable/models/#Inverter-Models

Finally, if you plan to use a lot the tool, I would recommend joining our Sienna Slack: https://join.slack.com/t/nrel-sienna/shared_invite/zt-glam9vdu-o8A9TwZTZqqNTKHa7q3BpQ

You can ask questions and share code more easily in the PowerSimulationsDynamics channel.