Sienna-Platform/PowerSimulationsDynamics.jl #398: Run test fails on test_case59_st6b and test_case57_degov

yasirroni, 2025-04-23:
**Describe the bug**
Run test fails on `test_case59_st6b` and `test_case57_degov`

**To Reproduce**
```julia
using Pkg

Pkg.activate(".")
Pkg.instantiate()
Pkg.precompile()
Pkg.test()
```

Below some output that might be useful to fix,

1. `sim.results` is `Nothing`

    ```text
    Test 59 ST6B ResidualModel: Error During Test at .../PowerSimulationsDynamics.jl/test/test_case59_st6b.jl:15
      Got exception outside of a @test
      State Vf not found in device generator-102-1. Available states: eq_p, ed_p, ψ_kd, ψ_kq, δ, ω, Vm, x_i, x_d, Vg
    
    Test 9 ST6B MassMatrixModel: Error During Test at .../PowerSimulationsDynamics.jl/test/test_case59_st6b.jl:87
      Got exception outside of a @test
      MethodError: no method matching get_voltage_magnitude_series(::Nothing, ::Int64)
      The function `get_voltage_magnitude_series` exists, but no method is defined for this combination of argument types.
    
    Test 57 DEGOV MassMatrixModel: Error During Test at .../PowerSimulationsDynamics.jl/test/test_case57_degov.jl:18
      Got exception outside of a @test
      MethodError: no method matching get_frequency_series(::Nothing, ::String)
      The function `get_frequency_series` exists, but no method is defined for this combination of argument types.
    ```

1. Evaluation is fail

    ```text
    Test 57 DEGOV MassMatrixModel: Test Failed at /Users/macbookair/Documents/Git/PowerSimulationsDynamics.jl/test/test_case57_degov.jl:73
      Expression: execute!(sim, MethodOfSteps(Rodas4(; autodiff = false)); dtmax = 1 / 240, saveat = 1 / 240) == PSID.SIMULATION_FINALIZED
       Evaluated: PowerSimulationsDynamics.SIMULATION_FAILED == PowerSimulationsDynamics.SIMULATION_FINALIZED
    ```
**Expected behavior**
All test passed

**Desktop (please complete the following information):**
 - OS: iOS - M1
 - Version latest on GitHub

**Additional context**
`unique_timestamps` on required input function also cause error, but fixed in separate PR.

jd-lara, 2025-04-23:
This is strange, I don't see why the tests would have broken I think this might be due to updates to DiffEq.jl

yasirroni, 2025-04-24:
It might be. I think that's because the packages in `Project.toml` is not complete enough, thus simply installing dependency to run the tests will update the `Project.toml`.

yasirroni, 2025-04-24:
Do you know when is the last working `DiffEq.jl`?

Should we restrict to make test past or should we fix the test to be compatible to the latest?

jd-lara, 2025-04-29:
This is weird, nothing has changed on those objects to induce such an error.

rodrigomha, 2025-05-01:
I'm sure that the issue on the test 59 is a bug in DiffEqBase when we apply a discrete callback. We should disable the test and investigate more if we can make a minimum example test to replicate the problem. @m-bossart we can probably check what the hell is going on.

rodrigomha, 2025-05-02:
I opened an issue in DelayDiffEq that should handle our test case57: https://github.com/SciML/DelayDiffEq.jl/issues/307

closed 2025-07-09 by jd-lara